greiman/SSD1306Ascii #119: Display the character °

pierrotm777, 2024-03-17:
Thanks for your nice library :-)

How can i print a ° character ?

Thanks

Pierre

greiman, 2024-03-17:
> How can i print a ° character ?

Most fonts only has standard ASCII characters.  I think [System5x7 ](https://github.com/greiman/SSD1306Ascii/blob/c43eaa9db195d17277ffc3d1ec11e3b21ba37775/src/fonts/System5x7.h#L144)has a degree symbol.  Maybe [Adafruit5x7](https://github.com/greiman/SSD1306Ascii/blob/c43eaa9db195d17277ffc3d1ec11e3b21ba37775/src/fonts/Adafruit5x7.h#L7) if ADAFRUIT_ASCII96 is defined zero.  I can't really help.

A number of people have generated special characters with openGLCDFontCreator.  I have never used this program.

pierrotm777, 2024-03-17:
Infact, System5x7 has a degree character.
It's the last.
I have used oled.print(char(128)).

Thanks,

Pierre