greiman/SSD1306Ascii #120: ShutDown OLED 1306

error105, 2024-05-17:
Hi, there is any option to turn off OLED to save power?

greiman, 2024-05-17:
There is no function to turn off the OLED.  `ssd1306WriteCmd()` is public so you could try sending` SSD1306_DISPLAYOFF` and `SSD1306_DISPLAYON`.

```
  oled.ssd1306WriteCmd(SSD1306_DISPLAYOFF);
  ...
  oled.ssd1306WriteCmd(SSD1306_DISPLAYON));
```
You can try it but I don't know if will save power or work properly.

deladriere, 2024-05-17:
I have  developed a bit of code to avoid the OLED fading off when displaying the same text
Here, I am just diming it, but you can also turn it off with the level of contrast

````
void setDisplayDIM()
{
        display.ssd1306WriteCmd(SSD1306_SETPRECHARGE);
        display.ssd1306WriteCmd(0);
        display.ssd1306WriteCmd(SSD1306_SETVCOMDETECT);
        display.ssd1306WriteCmd(0);
        display.ssd1306WriteCmd(SSD1306_SETCONTRAST);
        display.ssd1306WriteCmd(10);
}
void setDisplayNORMAL()
{
        display.ssd1306WriteCmd(SSD1306_SETPRECHARGE);
        display.ssd1306WriteCmd(0xF1);
        display.ssd1306WriteCmd(SSD1306_SETVCOMDETECT);
        display.ssd1306WriteCmd(0x20);
        display.ssd1306WriteCmd(SSD1306_SETCONTRAST);
        display.ssd1306WriteCmd(0xAF);
      }

```

noeldum, 2025-01-21:
I tested this.
On normal operation my hardware uses around 25mA.
If I run `oled.clear()` it goes down to 24mA.
If I run `oled.ssd1306WriteCmd(SSD1306_DISPLAYOFF);` I also get 24mA.

So basically the last command does not save more power than oled.clear().

oled.clear() can help not burning the screen too much but does not really save power.

Should I have a small mosfet controlling power to the oled display and activate/disable it from the mcu. Would this make a good deal of a difference?