greiman/SSD1306Ascii #122: Display is messed up

fitaine, 2024-09-15:
Hello,

I'm trying to build a [Servo operated 3D Printed Shutter for Arduino Nano](https://github.com/hackaninstant/Servo-Shutter) and I'm having trouble with the display when I use SSD1306Ascii.

My script wouldn't work so I tried to load the examples but non worked. Then, I tried to load examples from Adafruit SSD1306 library to know if the problem come from the OLED screen and it runs smoothly.

Here is the symptom: Display seems to be not aligned
1 - Wire 128x64 example 
[image]

2 - HelloWorldWire 128x64. I just get the last character of the oled.print ( here the exclamation mark) on the top left corner
[image]

3 - ssd1306_128x64_i2c (Adafruit SSD 1306 example)
[image]

The components I use are Arduino Nano and Oled screen 1306 from Aliexpress [link to product page](https://fr.aliexpress.com/item/32643950109.html?spm=a2g0o.order_list.order_list_main.75.6b835e5bheGidR&gatewayAdapt=glo2fra)

do you have any idea what I did wrong here ? Can you help me sorting it out please ?

greiman, 2024-09-15:
Looks like the display controller for this module is a 12864.  It claims to be 1306 compatible but I must use some command that is not handled by the 12864.

I am only doing minimal maintenance on this library so don't plan to add features or support new controllers.

fitaine, 2024-09-15:
Ok, so it might be a hardware compatibility after all. I've checked different product pages and I don't find the display controller info on it.
Would you mind pointing a 128x64 oled screen that would be compatible ?

Anyway, thanks for the clarification and the help ! I'm gonna buy an other one and try if it works better.

greiman, 2024-09-15:
> Would you mind pointing a 128x64 oled screen that would be compatible ?

I really can't help.  I developed this library nine years ago and it has been about five years since I bought an oled.  The ones I bought from Adafruit are now discontinued.

closed 2024-09-15 by fitaine