greiman/SSD1306Ascii #123: New SSD1306 128x64 I2C modules producing garbled text but work fine with Adafruit library

peteDDD, 2024-12-16:
I have been using this great library for at least 5 years now.  I really appreciate the small size and the great feature set which is much better for managing constantly changing ascii screens.  Thanks!

Now for the last few batches of SSD1306 .96-inch 128x64 modules, they are all displaying garbled text.  In fact, I threw out a bunch of them thinking they were bad from the factory...   I decided though to test them using the example Adafruit library code and they worked just fine.  No garbled text.  The thing is, I prefer your library as I have built many of your functions into my code that the Adafruit library just does not have.  I have tried multiple fonts and two I2C speeds but get similar results.

I am using your example code for the test.
Also please see attached photos of the display output from this code.

Again, this is the second batch of displays from different manufacturers that have shown these problems.
This one is labeled: "TS00356a-0.96-Lanban-Bai   Display Module 12864 LCD Screen Board for Arduino Color: SSD1306 White"

Any clues for me?

THANKS!

```
#include <Arduino.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

// 0X3C+SA0 - 0x3C or 0x3D
#define OLED_I2C_ADDRESS 0x3C

// Initialize display
SSD1306AsciiWire oled;

// forward declaration
void clearField(uint8_t col, uint8_t row, uint8_t width);

void setup()
{
  Wire.begin();
  // two clock speeds tried
  Wire.setClock(400000L); // Fast I2C mode
  // Wire.setClock(100000L); // Slow I2C mode

  Serial.begin(9600);

  // Initialize OLED - try these options one at a time:
  oled.begin(&Adafruit128x64, OLED_I2C_ADDRESS); // Current - produces garbled characters   
  // oled.begin(&SH1106_128x64, OLED_I2C_ADDRESS);        // Try this option for SH1106 displays
  // Alternative options if needed:
  // oled.begin(&Adafruit128x32, OLED_I2C_ADDRESS);    // For 128x32 displays
  // oled.begin(&Adafruit64x32, OLED_I2C_ADDRESS);     // For 64x32 displays

  // Set font  Three fonts tried
  // oled.setFont(Callibri15);
  // oled.setFont(System5x7);
  oled.setFont(Adafruit5x7);

  // Clear the display
  oled.clear();

  // First row - double size text
  oled.set2X(); // Double size text
  oled.println("Hello!");

  delay(2000);

  // Second row - normal size text
  oled.set1X(); // Normal size text
  oled.println("OLED Display");

  // Third row
  oled.println("Testing 123");
}

void loop()
{
  // Your loop code here8X64
}
```

Platformio.ini:

```
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
;upload_port = COM4
board_build.variants_dir = variants
board_build.variant = mcupro
lib_ldf_mode = chain
lib_extra_dirs = ~/Documents/Arduino/libraries

lib_deps = 
    greiman/SSD1306Ascii @ ^1.3.5
```

greiman, 2024-12-16:
> Any clues for me?

I suspect the controller is not strictly SSD1306 compatible.

Unfortunately I no longer have time to maintain this library.

peteDDD, 2024-12-16:
ok.  but thanks for what was a great library!

greiman, 2025-09-22:
[tomahawk66](https://github.com/tomahawk66) There are now several new variations on the SSD1306 in recent OLED modules.

I need to acquire samples in order to address this issue. Currently I am very busy so don't know when I will update this library for these controllers.

tomahawk66, 2025-12-28:
I have sent you a couple of these modules with a friend traveling to the USA, who will mail them to you locally. Many thanks!

greiman, 2026-01-04:
I received the two modules.  They appear to have a NFP1315-61AY controller.  I assume this is some form of an SSD1315 controller which is nearly compatible with a SSD1306.

I have tried my library, the Adafruit SSD1306 library, and the u8g2 library.  I setup an example that work on my test module and then try your module.  I can't get anything to display.  The u8g2 library has many options and none of the ones I tried worked.

Here is a photo of one of your modules on the left and one I use for tests on the right. My module is a 128x64 with a SSD1306 compatible controller.

[image]

The seven pins on the two modules have Identical labels.  The circuit boards look identical with the same value resistors.

The big difference is the ribbon cable.  Many wires are missing in your modules.

I am not sure what to try next.  If I could get an example to work with another library, I could explore the differences.

tomahawk66, 2026-01-04:
Thank you for looking into this.

As above, I did get them to work using the Adafruit library.

20260104_092715.jpg (view on web)<https://github.com/user-attachments/assets/530e5867-1a68-4dc7-93b9-1468d43526e5>