greiman/SSD1306Ascii #125: How to print "°C" on OLED?

mianqi2016, 2025-03-22:
How to print the Celsius degree sign?

greiman, 2025-03-22:
Like this:

```
void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  Serial.println(u8"How to print °C using UTF-8.");
}
void loop() {}
```

greiman, 2025-03-22:
Most fonts only support ASCII and don't have the degree symbol. I think the Adafruit5x7.h font may have it if you set [ADAFRUIT_ASCII96](https://github.com/greiman/SSD1306Ascii/blob/c43eaa9db195d17277ffc3d1ec11e3b21ba37775/src/fonts/Adafruit5x7.h#L7) to zero.

Also the last character in [System5x7.h](https://github.com/greiman/SSD1306Ascii/blob/c43eaa9db195d17277ffc3d1ec11e3b21ba37775/src/fonts/System5x7.h#L144) is the degree symbol.

Other users have created fonts with special symbols using [openGLCDFontCreator](https://github.com/greiman/SSD1306Ascii/tree/master/extras).

mianqi2016, 2025-03-25:
> the last character in [System5x7.h](https://github.com/greiman/SSD1306Ascii/blob/c43eaa9db195d17277ffc3d1ec11e3b21ba37775/src/fonts/System5x7.h#L144) is the degree  symbol.
>

How to use it? 
These code didn't work:
`  oled.setFont(System5x7);
  oled.println("\127C");`