greiman/SSD1306Ascii #126: Font Creator is not able to open font files?

Miq1, 2025-05-29:
> I have to pick up the thread again here - I am trying to edit the System5x7 font without luck.
> I downloaded the Font Creator linked above, but when trying to open the System5x7.h file with it I get an error message only:
> ```
> Excpetion: java.io.StreamCorruptedException: invalid stream header: 2F2A0A20
> ``` 
> It looks like the Font Creator is looking for a different kind of font file? 

 _Originally posted by @Miq1 in [#85](https://github.com/greiman/SSD1306Ascii/issues/85#issuecomment-2918714337)_

greiman, 2025-05-29:
Looks like the original GLCD font creator app is no longer supported.

The [site ](https://www.mikroe.com/glcd-font-creator)I downloaded the app from is down.

I don't know how to use any other app to edit/create fonts.

The version [here](https://github.com/greiman/SSD1306Ascii/tree/master/extras) is very old.

Miq1, 2025-05-29:
Hi Bill, thanks a lot for the quick response.

I tried the version from the 'Extras' section as well, but  to no avail. Looks like I will have to edit the header file the old way - by hand. 😉