greiman/SSD1306Ascii #127: error: #error AvrI2c not supported for this chip ESP8266 with or without onboard OLED display using I2C interface

valveman2006, 2025-08-10:
ESP8266  with or without onboard OLED display using I2C interface

#include <SSD1306AsciiAvrI2c.h>  on compilation gives the following compiler error

#error AvrI2c not supported for this chip.
 

// ----------------  sample test code  -------------------------

#include <SSD1306Ascii.h>
#include <SSD1306AsciiAvrI2c.h> 
#include <SSD1306AsciiSoftSpi.h>
#include <SSD1306AsciiSpi.h>
#include <SSD1306AsciiWire.h>
#include <SSD1306init.h>

//------------------------------------------------------------------------------
void setup() {                
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
  oled.print("Hello world!");
}
//------------------------------------------------------------------------------
void loop() {}

greiman, 2025-08-10:
Yes, ESP8266 is not an AVR board.

Try [this](https://github.com/greiman/SSD1306Ascii/tree/master/examples/Wire128x64) example for I2C on ESP8266.

valveman2006, 2025-08-10:
Many thanks there is a lot of misleading info on the web about your library, I have used it for years successfully with the different Arduino versions. Any possibility of a version for the ESP chips.